westlicht/performer #105: Metropolis / Ryk M-185 style sequencing

madeinspace, 2019-07-21:
Just dreaming out loud but I would love to see a sequencer replicating the functionality of the Metropolis/Ryk M-185  on multiple tracks. that would involve:

1: pulse/repeat count up to 8 or more per step
2: skip or glide per step
3: gate mode (hold, repeat, single or rest) per step

great demo here: https://www.youtube.com/watch?v=uV9-XA5MPwY&feature=youtu.be&t=152

westlicht, 2019-07-21:
I have a development branch with a _stage_ track mode which basically will allow this kind of sequencing. But its not on top of the priority list right now.

madeinspace, 2019-07-22:
OMG! :) I'll follow closely

toneburst, 2019-07-27:
Looking forward to this, too :)

dComposer, 2019-08-05:
I was thinking about this mode a bit today: it would be really awesome if you could hold down a S1-S16 button while twisting the encoder to change the pitch in real time. That would definitely give this mode a fun metropolis feel!

jules, 2019-08-28:
Would love to this feature implement and yes, for the S1-S16 button for real time changes !

jstefani, 2019-10-09:
(thinking out loud for the day when we do max out the memory of the machine)

If the problem is memory (I believed mentioned in another issue), perhaps we could eventually repurposes other pattern tracks (is there a name for them? "auxiliary lanes" perhaps? For example, note probability, etc. 

It could quickly become convoluted and confusing but what if instead of saying :

each track has gate, gate probability, gate offset, repeat, repeat probability, note, note change amount, etc...

We just said

each track has 8 (or however many) auxiliary lanes which the user may assign to various functions.

problems might be:

(i would guess) all the current "auxiliary lanes" do not store the same datatype so cannot be easily interchanged

ui setup

how stored? per project i guess.

general confusion

theotteryears, 2020-05-02:
I would certainly like this feature too

franq87, 2020-07-10:
any news about this? I'm super interested on this feature, it would be really great having this on the performer

gitmusik, 2020-07-19:
Me, too!

dragosandriana, 2020-08-19:
PERFORMER is the best and most complex modular sequencer.
But this feature is a MAJOR miss!
When Performer will have this feature there will be absolutely no rival ever!
Following this, hoping it will become a priority.

kooops, 2020-10-26:
+1
Little UP
Same as anybody else: THANK YOU for the performer!

theotteryears, 2020-11-23:
+1 (would very much like this feature)

hotelbachelard, 2020-11-26:
I just signed in to Github for the first time to +1 on this.

vafu, 2020-12-05:
Hi! I've been making some attempts to implement this, for now I have a draft PR, so anyone eager to help testing and making UX decisions -- please contact me! 
Here's the PR: #259

Currently added step repeats and repeat modes, it brings closer to the solution.
All these parameters are currently hidden under GATE page. Metropolis is an amazing module because of its immediacy so I'm thinking about providing access to these pages by SHIFT + fn button, for instance for step repeats it's SHIFT + RETRIG, for gate modes its SHIFT + GATE.
Let's discuss!

hotelbachelard, 2020-12-05:
> Hi! I've been making some attempts to implement this, for now I have a draft PR, so anyone eager to help testing and making UX decisions please try it!
> Here's the PR: #259
> 
> **THIS IS AND PROBABLY WILL BE WORKING ONLY IN FREE PLAY MODE** due to nature of Aligned mode which always snaps to absolute steps grid.
> Currently added step repeats and repeat modes, it brings us closer!
> My concern is that all these parameters are currently hidden under GATE page, so you have to click over and over again to flip between them. Metropolis itself is an amazing module because of its immediacy so I'm thinking about providing access to these pages by SHIFT + fn button, for instance for step repeats it's SHIFT + RETRIG, for gate modes its SHIFT + GATE.
> Let's discuss!

I personally don't mind that it's under the Gate page, if it's an extra page to get to "Repeat" settings. I can live with the limitations of the screen, which is already bigger than most screens on Eurorack modules. 
Quick question - does this step repeat appear on Curve tracks as well? I imagine that would duplicate the curve shape, but i wonder how that might mess up the Curve track's own Gate settings.

vafu, 2020-12-05:
@hotelbachelard no, this is working only for note track, implementing this to curve tracks is whole different story :)