westlicht/performer #107: Elder project not loading

SloBloLabs, 2019-07-22:
Took me a while to step to 1.26, but today was the day. Unfortunately, some project are not accepted to load and the sequencer stays reset playing empty sequences.
Sorry to say I can’t tell exactly the version used to write them. But I could provide the project files from SD card if of any interest.

westlicht, 2019-07-22:
0.1.6 did break backwards compatibility. If you used a later version, send me the project files and I can have a look. In general they should load fine but there might be small incompatibilities I havent caught.

closed 2019-07-22 by westlicht

SloBloLabs, 2019-07-23:
Yes, sure. What's your preferred method to receive the file?

westlicht, 2019-07-24:
You can just drag a file to a new comment.

SloBloLabs, 2019-07-24:
Version 1.26
Project 1 and 2 loading properly, 3 and 4 determined as invalid project files, 5 causing module reset.

[PROJECTS.zip](https://github.com/westlicht/performer/files/3427826/PROJECTS.zip)

westlicht, 2019-07-24:
Had a quick look at the project files and this is what I read from the files:

001 - file version 4 (v0.1.8)
002 - file version 4 (v0.1.8)
003 - file version 11 (v0.1.21)
004 - file version 11 (v0.1.21)
005 - file version 11 (v0.1.21)

Its surprising that the files written with v0.1.21 are broken. This should not be the case. Did you use an official release when writing these files or was your firmware compiled from source (e.g. master branch)?

SloBloLabs, 2019-07-24:
Thanx for investigating. Yes, I do my own compiles. Had no issues in that are so far. Can you confirm loading doesn’t work on your end neither? If steppin back to 1.21: what would be a workable migration path?

westlicht, 2019-07-24:
Yes, same result using even the v0.1.21 version over here. Compiles from master can be dangerous when it comes to file versioning. The reason is that whenever a new data field is added to the data model, the file version is incremented. However, when I add multiple new fields from one release to the next, I keep adding to the same file version, meaning that writing files in between releases sometimes would only work on that exact commit. It might make sense to do this differently in the future and possibly increment the file version multiple times between releases.

As for how to mitigate this, it's a bit tricky. You would have to figure out which exact commit the project files were written from, then check what has been added before the next release. Then hack the firmware to read the "inbetween" version but write a correct "release" version. I'm not willing to do that for you, but feel free to try yourself. And in the future, it makes sense to stick to release versions.

SloBloLabs, 2019-07-24:
Sure, get that point. Well, the files didn’t contain enough “gold” to put effort in that anyway, haha. Lessons learned, I’ll stick to official release when working on and saving next projects ;-)

westlicht, 2019-07-24:
Oh I didn't know these projects mean that much to you. Sorry to hear. Do you feel like you have the ability to do this yourself? I can have a quick look, if there was not too much noise in between those releases it might be a quick one.

SloBloLabs, 2019-07-24:
No, I wanted to express they don’t have big value to me. Everything sorted, Simon. Thank you so much!

SloBloLabs, 2019-07-24:
Sorry for confusion. ;-)

westlicht, 2019-07-24:
Ah i see :) completely mis-read that statement. I actually think I know where the thing broke. I give it a try to just make sure no funky business is going on and my analysis was correct.

westlicht, 2019-07-24:
Ok, it was all as I expected. Here are the fixed project files.

[PROJECTS.zip](https://github.com/westlicht/performer/files/3428510/PROJECTS.zip)

SloBloLabs, 2019-07-24:
Wow, great, thank you 1000 times. Of course now I am curious what did the trick?

westlicht, 2019-07-24:
Looking at https://github.com/westlicht/performer/compare/v0.1.20...v0.1.21 you can see that two fields were added to the model (_clockOutputSwing_ and _curveCvInput_). You used a commit that did write the _clockOutputSwing_ but not the _curveCvInput_ yet, but they use the same project file version. So when you try to read back in a later firmware version the loaded expects both fields to be there but in the files only one is present.