westlicht/performer #108: Curve sequence improvements

westlicht, 2019-07-22:


dComposer, 2019-07-23:
I love that you have the curve tracks because if I'm only using 2-3 tracks for a song then that gives me a bunch of LFO's to play around with on the leftover tracks. Maybe an extra "curve" type could be a random s&h?

westlicht, 2019-07-23:
@dComposer if you want random s&h like from a curve track, this is pretty simple to achieve. Select all steps and use the generator to randomize the minimum or maximum value. Then either let it run in periodic (forward/backward/pingpong etc.) or use one of the random modes to advance the sequence for even more randomness. Once you master the controls of the sequencer this can literally be done in a few seconds. No need for a s&h curve type.

westlicht, 2019-07-24:
Will be part of 0.1.27 release.

commit d318f3de (2019-07-24): improved curve track

- added shape variations with probabilities
- added gates with probabilities
- added shape and gate probability bias
- better support for track linking
- support for fills (variation, next pattern, invert)
- support for mutes

fixes #108
closed 2019-07-24 by westlicht