westlicht/performer #109: Additional arpeggiator modes

dComposer, 2019-07-23:
I just played a show where I relied heavily on the arpeggiator and I found myself really wishing for an Up Down mode that went:

1 3 5 | 1 3 5 | 1 3 5 | 1 3 5 | 5 3 1 | 5 3 1 | 5 3 1 | 5 3 1
     oct1   | oct2 | oct3  | oct4 | oct4  | oct3 | oct2 | oct 1

and, conversely, a Down Up mode that went:

5 3 1 | 5 3 1 | 5 3 1 | 5 3 1 | 1 3 5 | 1 3 5 | 1 3 5 | 1 3 5
     oct4  | oct3 |  oct2 | oct1  |  oct1  | oct2 | oct3  | oct4 

(sorry for the lousy ascii artwork!)

westlicht, 2019-07-31:
Changed _Octaves_ parameter to allow for _Off_, _Up 1-5_, _Up Down 1-5_, _Down 1-5_ and _Down Up 1-5_. When moving down octaves, the note order is reversed. This basically allows to do what you suggested and a bit more. Note however that the sequence is always started at octave 0. So your second example would start at octave 0 and move down to octave -3. But this is more natural as you expect the sequence to start from where you play on the keyboard.

Will be part of the 0.1.30 release.

closed 2019-07-31 by westlicht

dComposer, 2019-08-15:
Just wanted to let you know that I had a ton of fun playing with the new arp modes tonight! They sound incredible!!!

commit 696b7af7 (2019-07-31): improved arpeggiator octave patterns

fixes #109