westlicht/performer #110: Enhanced midi routing

dComposer, 2019-07-23:
The midi functionality of performer was the brain of my last performance and I used the midi tracks extensively. I would love it if a few of these midi routing enhancements would be implemented:

- midi split: all notes up to some arbitrary note get sent to one cv/gate and all notes past that point get sent to another cv/gate combo

- per note routing: each note gets sent to a separate cv/gate combo (ie, for a root position C major chord: C1 is sent to track1, E1 is sent to track2, G1 is sent to track3, C2 is sent to track4). I'm currently fantasizing about sending each note to a different voice! It could have 2 options for the routing: play order or chord order. Play order would mean that if I played the E first, it would go to track1, the G second it would go to track2, etc; while chord order would mean the bottom note always goes to track1, the next higher note always goes to track2, etc.

- midi filtering: If I'm holding a C major chord down, I'd love it if I could send the bottom C to one track and the full C chord to another track with an arpeggiator set up. I could set this up in the midi track settings by telling track1 to filter out all notes except for the bottom note I'm holding down and transpose it down an octave, and then set up track2 to use the same midi information but command it to use the arpeggiator (for example).

Relatedly: are the arpeggiator options cv-routable? Like, can I turn the arpeggiator on/off by sending a trigger, or change the arp divisions using an offset generator sent to a cv input?

Thank you for everything! This module is amazing!!

westlicht, 2019-07-23:
> midi split: all notes up to some arbitrary note get sent to one cv/gate and all notes past that point get sent to another cv/gate combo

I will consider this. Adding a _key range_ to the MIDI/CV track seems like a nice addition allowing for setting up a split keyboard patch.

> per note routing: each note gets sent to a separate cv/gate combo (ie, for a root position C major chord: C1 is sent to track1, E1 is sent to track2, G1 is sent to track3, C2 is sent to track4). I'm [...]

> midi filtering: If I'm holding a C major chord down, I'd love it if I could send the bottom C to one track and the full C chord to another track with an arpeggiator set up. I could set this up in [...]

This is not something I will implement. The MIDI/CV track was primarily intended to allow using the sequencer as a MIDI/CV converter. I will probably add MPE in the near future as soon as I have a capable device. But this is more in the territory of a MIDI processor. I would suggest you setup your tracks on different MIDI channels and use an external MIDI processor to do these.

> Relatedly: are the arpeggiator options cv-routable? Like, can I turn the arpeggiator on/off by sending a trigger, or change the arp divisions using an offset generator sent to a cv input?

Currently these parameters are not routable. I might make some of them routable in the future. At least the divisor, as there is a direct correspondence to the other track modes.

westlicht, 2019-07-24:
Will be part of the 0.1.27 release.

closed 2019-07-24 by westlicht

UncleRoel, 2019-07-24:
A way to partly implement this idea is to implement a choice for a note stealing algorithm. (Lowest note, highest note, first note, last note. ) 
This is very common in midi cv conveyors

westlicht, 2019-07-24:
I don't like to have "communication" between different tracks. Maybe you can elaborate how you would envision this idea.

dComposer, 2019-07-24:
Oh that's simple! I envisioned it being like:

- Set Track 1 to Midi Ch1 and have it only read the lowest note

- Set Track 2 to Midi Ch1 and have it only read the second note

- Set Track 3 to Midi Ch1 and have it only read the third note 

- Set Track 4 to Midi Ch1 and have it only read the highest note

etc etc

...So each track is set to filter a specific note that you're playing in a chord.

PS: Thanks for implementing the key range idea!

westlicht, 2019-07-24:
The problem I see with this is the following: When playing a chord, you don't necessarily play the notes in order (low to high). So the "second note" might change during playing a chord. Also, to detect a chord you would have to wait some time to get all notes in to do the decision and dicing it to different tracks. So there is quite some logic involved to get this going unless I completely miss the point.

UncleRoel, 2019-07-24:
The note stealing algorithm is not about different tracks. It was implemented by design in old mono synths, and the basis of a typical playing technique: you hold down a low note, then play short notes above, and in between the low note sounds as it's all playing (here note priority is highest note). Quite like a tapping on guitar. 

The way this can be used is having different channels with different note priorities/stealing algorithms listening to the same midi input. 

I think your midi cv always uses the last played note now, and doesn't keep a stack to go back to the previous note after the last one is released? I should test it.

dComposer, 2019-07-25:
Yeah, I was envisioning something really dumb with no logic involved: "Lowest" just send the lowest note of a chord to whatever track you have that set up for, "Highest" filters for the highest chord, etc. It would be really handy for the arpeggiator because if I select "lowest" I could send root notes to other tracks, or if filtering for "highest", a melody line that is supported by an arpeggiator pattern.

toneburst, 2019-07-27:
+1 for the "note-stealing" options (if not already implemented).

westlicht, 2019-07-28:
Opened new issue #115 for implementing note priority modes.

commit 431d4bc2 (2019-07-24): implemented key range for midi cv track

this allows for split keyboard between two midi cv tracks

fixes #110