westlicht/performer #111: Song mode.

Pipstapee, 2019-07-25:
Is it possible to increase the song mode to more than 16 parts, 64 active parts using the page system?

westlicht, 2019-07-25:
Did you reach the limit?

Pipstapee, 2019-07-25:
I like to lay out a song in, intro,Drop,Main,Mid and outtro. Is the limit 16?

westlicht, 2019-07-25:
Currently yes its 16 slots.

dComposer, 2019-07-25:
Perhaps something to chew on for down the road: I feel like your performer screensaver (the one with all 16 tracks on the y axis and the steps on the x axis) would make a more intuitive version of song mode. Each cube could be a pattern so you weren't stuck with all tracks having to play the same pattern. Initially that's what I thought this screen was for.

westlicht, 2019-07-25:
Every song slot can have an arbitrary set of patterns. Read the manual on how to do that. I was initially looking for better visualizations but couldnt get it into the required space/pixel count.

westlicht, 2019-07-31:
moved of #118

closed 2019-07-31 by westlicht