westlicht/performer #112: Glitchy encoder

sunflowr, 2019-07-25:
Asked in the pusherman discord and others seemed to also experience this as well.
When pressing the encoder it sometime also register as if I'm turning it to the right meaning I some time jump to the wrong parameter or even change the value by accident.

sunflowr, 2019-07-25:
Tested doing a super small movement of rotating the encoder to the right until I feel the dent without actually going pass it and it register it as a rotating one step.

Not sure if this is intended behaviour or not but I was expecting it to only register when going pass the dent.

av500, 2019-07-25:
from looking at the code, I think it does not take the 4 transitions between encoder A and B states into account. ideally there is a 4 step counter or 4 states and only when the counter counted +4 or -4 a step should be emitted. then you also dont need to do the debouncing...
see e.g. here: http://bytes.usc.edu/ee109/lab-7/

westlicht, 2019-07-25:
Hopefully fixed encoder handling once and for all :D This will be part of the 0.1.28 release.

closed 2019-07-25 by westlicht

toneburst, 2019-07-25:
Great!

commit 77642afc (2019-07-25): implemented better rotary encoder scanning

this is now based on a state table decoding the gray code. no longer requires debouncing.

fixes #112