westlicht/performer #114: shifting sequence (first/last step idea)

cornil-m, 2019-07-28:
Hey,
First of all thanks for your work :) 

it could be nice to shift a sequence without changing the number of step.

I was thinking maybe it could be possible to shift the sequence  if the first or last step selection is out of the previous region.
For example you have first step at 1 and last step at 4, if first step select is bigger than last step, then the whole seq could switch, in this example selecting step 5 as first step would shift the seq from 5 to 9. then selecting 4 as last step would send back to 1-4.

cornil-m, 2019-07-28:
or maybe using shift while selecting first or last step could be easier actually

westlicht, 2019-07-28:
Good idea. I have implemented the following:
- On the sequence edit page, if first/last step parameter is edited with `SHIFT` being hold, both first/last step are changed in sync.
- On the step edit page, if first/last step parameter is edited (quick access) with `ENCODER` being pressed, both first/last step are changed in sync.

Will be part of the 0.1.28 release.

commit bac74c1e (2019-07-28): allow first/last step to be moved together

- when holding `SHIFT` on the sequence edit page
- when holding the `ENCODER` on the step edit page

fixes #114
closed 2019-07-28 by westlicht