westlicht/performer #116: More Selection options

soporus, 2019-07-30:
As I try and fail with entering notes via live record / keyboard,  I’m dreaming a lot about more selection tools to perform sequence surgery.  Or any of these plausible in the future? 

- Record and select:  all newly recorded steps are recorded in the persistent mode.  This would be handy for recording in multiple passes.  Record one pass with a certain trig conditions, and the next with different, and so on.

- Record within selection only:  would help with adding little touches to an existing sequence, without removing your otherwise perfect take.

- Select only On or Off steps.

- Generators: “generate and select” or “generate as selection only” modes.  This one maybe overlaps with that series selection key combo.

westlicht, 2019-07-31:
> Record and select: all newly recorded steps are selected in the persistent mode. This would be handy for recording in multiple passes. Record one pass with a certain trig conditions, and the next with different, and so on.

What is a pass? When is next? I think it might be a valuable addition but not high priority for me.

> Record within selection only: would help with adding little touches to an existing sequence, without removing your otherwise perfect take.

Yet another mode. Not to fond of this. There is plenty of similar things you can already do. Route a pedal or MIDI controller to record to allow punch in/out on the fly (I used this quite a lot before). Use snapshots to quickly create a temporary copy that you can mess with and go back if you screwed up.

> Select only On or Off steps.

This is on my mental todo list. My idea was to have `SHIFT` + double tap a step select all steps that have the same value on the currently selected layer.

> Generators: “generate and select” or “generate as selection only” modes. This one maybe overlaps with that series selection key combo.

Will keep in mind when I redesign the generator system and add new modes.

soporus, 2019-07-31:
> What is a pass? When is next? I think it might be a valuable addition but not high priority for me.

A pass could be all new writes to steps since record was engaged.  Leaving record would end the pass.  Maybe _Page + Shift + Rec_ to engage this, since shift is already associated with selection?  There is the issue of how to handle existing selections though. My preference would be that engaging this mode would drop existing selections in the track.  The alternative seems to be adding to existing selection, but not sure that could be as useful.

> > Select only On or Off steps.
> 
> This is on my mental todo list. My idea was to have `SHIFT` + double tap a step select all steps that have the same value on the currently selected layer.

That sounds like a clean / clear way to implement it, I like that a lot.

westlicht, 2019-07-31:
I will move this to the backlog as it adds quite a bit of complexity where other workflows are already in place that allow you to do similar things (except for selecting the newly recorded notes).

closed 2019-07-31 by westlicht