westlicht/performer #118: Song page improvements

westlicht, 2019-07-31:
- extend to maybe 32 or 64 slots
- display slots in a table to allow seeing exactly what patterns are triggered on each slot
- add more visual playback information
- add function to duplicate a slot
- move clear function to context menu and rename to init (consistent with other pages)
- better visual feedback when assigning patterns to slots
- do not switch tracks when pressing track buttons (these are used to select patterns for specific tracks)

westlicht, 2019-08-11:
Will be released with 0.1.30.

closed 2019-08-11 by westlicht

dComposer, 2019-08-15:
Does the inclusion of the time signature mean we're not limited to only writing songs in 4/4 now??

westlicht, 2019-08-15:
Yes it does. Obviously sequences can still run at a different time division and length. All the measure/bar length are multiples of the time signature now.

commit 1234f9c7 (2019-08-11): refactored song page

- slots are displayed in a table
- editing slots using encoder and step buttons

fixes #118