westlicht/performer #121: MIDI-CV Mono Legato

toneburst, 2019-08-05:
Mono legato mode, with adjustable glide-time for single-voice MIDI to CV channel.

Gate is not retriggered on overlapping MIDI notes, and pitch CV glides to new value over selected time.

Staccato notes operate as normal.

toneburst, 2019-08-08:
Glide time could also respond to General-MIDI-standard MIDI CCs 5 (Portamento Time) and 68 (Legato mode On/Off).

westlicht, 2019-08-13:
This would be a good addition. But its a bit tricky, especially if also listening to the midi messages for controlling portamento as they would interfere with manual settings on the track page.

toneburst, 2019-08-13:
Fair points.

Maybe scrub CC control, then and have portamento time just honour the setting on the Track page.

You could add an option somewhere to turn mono legato/slide on for monophonic parts.

It would either not show up at all for polyphonic parts, or would be "greyed-out".

toneburst, 2019-08-13:
I don't see a Portamento setting on the Track page, for MIDI/CV track mode in the User Guide.

Do MIDI/CV tracks have a Routing page? If yes, perhaps portamento/slide-time could be routable (with eg a MIDI CC).

westlicht, 2019-08-14:
Added slide time setting to midi/cv track, same as for note and curve track. This is a routable parameter. Will be released with 0.1.31.

closed 2019-08-14 by westlicht