westlicht/performer #122: Special curve : noise

cornil-m, 2019-08-08:
Would it be possible to add noise as curves track? Some digital garbage could be great for fm duties in alternance with clean curves.

westlicht, 2019-08-12:
Not sure about this one. What would the frequency of the noise be? Curves are currently mathematical expressions that are evaluated on the fractional time of a step.

cornil-m, 2019-08-13:
What kind of mathematical expression is it? I’m not familiar with coding but maybe adding random vector could work.
https://en.m.wikipedia.org/wiki/White_noise

https://en.m.wikipedia.org/wiki/Multivariate_random_variable

UncleRoel, 2019-08-13:
An easy way would be just generating a random value at each fractional time. Depending on the length of these fractions, this will give you a certain type of noise...

dComposer, 2019-08-13:
oooh another flavor of this could be to add a random walk pattern so that the cv randomly increases or decreases based on t-1

westlicht, 2019-08-13:
The problem is that curves are stateless. The sequencer does not store the current value but simply evaluates an expression when a value is needed. So it also has no "resolution". Its not impossible to add this but I don't think it adds a lot of value.

toneburst, 2019-08-14:
> The problem is that curves are stateless. The sequencer does not store the current value but simply evaluates an expression when a value is needed. So it also has no "resolution". Its not impossible to add this but I don't think it adds a lot of value.

Is it not "cheaper" to store the curves in a lookup table, rather than generating values on-the-fly at the DAC sample-rate?

OT, I know, but I'm curious.

westlicht, 2019-08-14:
> Is it not "cheaper" to store the curves in a lookup table, rather than generating values on-the-fly at the DAC sample-rate?

These are simple expressions and the CPU is fast, so its fairly cheap. When you use lookup tables you probably have to do interpolation too, so in the end it's not much cheaper if not more expensive because of the memory lookups. But that's not the issue here. The issue is that sure, I can create a function that creates a pseudo random output at a specific frequency, but then it's always going to be the same for every step. Otherwise you have to add a seed value and keep track of that, something which all other curve shapes don't need. It complicates a otherwise perfectly simple design for a small feature that I'm not too crazy about myself. Maybe I need to try it to see if there is enough value. My goal is to create software that retains maintainability, which these kinds of "hacks" make more complicated. It's not that a single one of them makes things unmaintainable, but the sum of them is. At the current complexity of this firmware I have to be more and more resilient to just adding more stuff.

av500, 2019-08-14:
why not use the ADC to read an analog pin or even the internal voltage or temperature values  and derive a seed from that at boot?

UncleRoel, 2019-08-14:
Or even seed at the time you start generating random values. How do you calculate the chances for steps? Don't you have a random generator running there?

Also, HAL_GetTick() is a standard value for seeding random numbers.

westlicht, 2019-08-14:
All of those are not issues at all. I guess I didn't explain the issue well enough. If you're interested go study the implementation, it will make it clearer. Again, the point is not that I don't know how to implement this, but it requires turning a very clean simple straight-forward design into a less clean one, which I object unless there is a good reason for it. As mentioned above I'm getting more cautious with these kinds of trade-offs at this stage of the project.

froschmett, 2019-08-18:
Totally agree, there shouldn't be pink ponies with green dots wearing blue elephant t-shirts while surfing rainbows...

closed 2019-12-01 by westlicht