westlicht/performer #124: multiple midi tracks using the same midi channel?

dComposer, 2019-08-10:
Hi - I'm trying to set up a patch for a show tonight and for some reason I can't get two tracks to get notes from the same midi channel. Is this a bug or am I just doing something wrong? (I have track 1 set up so that it creates an arpeggio from midi ch 1 but I'm not getting anything out of track 3 which is set up to do the exact same thing). Thanks!

westlicht, 2019-08-11:
That is actually true, if an event is consumed it is not sent to other tracks. For MIDI/CV tracks this probably does not make sense. I will change it with the next release.

westlicht, 2019-08-11:
Hope you were still able to play the show!

westlicht, 2019-08-11:
Will be released with 0.1.30.

closed 2019-08-11 by westlicht

dComposer, 2019-08-12:
Thanks Simon! Yeah it didn't affect my show at all -- I was just trying to figure out an efficient way to switch arp patterns on the fly (I improvise scores to silent horror movies and hearing the same arp pattern for 90 minutes in a row gets a bit stale!) so I built Tesseract's Buff Matrix and I thought I'd be able to just cue up a bunch of different patterns in per|former and send the cv outs to the various channels of the Matrix to perform the switching duties when I ran into this bug. I ended up just switching the pattern to "play order" so I could control the pattern a bit and break the monotony. Really appreciate you fixing this so quickly! This will be really cool for layering a bunch of oscillators with different arp patterns now!

PS: Here's the performer in action last night on the Coney Island Freak Show stage:
[image]

commit 4542b447 (2019-08-11): allow multiple midi/cv tracks react to same messages

fixes #124