westlicht/performer #125: sending midi clock from midi out

indraperkasa, 2019-08-11:
hi Simon,

is it possible to implement send the master tempo bpm pulse thru midi out as a midi clock?
an example case, i would like to use my reverb/delay pedal that can accept midi clock to sync with the delay timing.
by the way really love this sequencer! thanks so much for doing this project.

Indra

westlicht, 2019-08-11:
Read the manual please. Midi clock is implemented (both in and out). Go to clock setup page to enable it.

westlicht, 2019-08-11:
And glad to hear you like the sequencer! Thanks!

closed 2019-08-11 by westlicht

indraperkasa, 2019-08-11:
ah sorry about that, i was reading the manual and couldn't find it, i guess i'm gonna re-read again. thanks so much!

westlicht, 2019-08-11:
Yep, the firmware is quite deep at this point so it's hard to really know all of it's corners. I should start on a video tutorials series because sometimes it feels people do not know some of the coolest workflows available :)

indraperkasa, 2019-08-11:
ok just found it, i guess i have skipped that midi clock info haha. 
yeah please do make some video tutorials, that would be really nice and helpful, indeed there's so many cool things that i just discovered recently, although been using it for a couple of months 👍 👍