westlicht/performer #126: Rudimentary S1-S16 keyboard for recording patterns

dComposer, 2019-08-13:
I was thinking the other night while programming a pattern with the encoder just how much quicker it would be if S1-S16 acted like a piano keyboard when I was in record mode (and I didn't have a midi controller handy). S9 would be C, S1 would be C#, S10 would be D, S2 would be D#, S11 would be E, etc. maybe the lights could be lit in the pattern of the keyboard to give some sort of visual:

[image]

westlicht, 2019-08-13:
I don't like the step keys to be reassigned for that by default. Having one octave only is also a bit limiting. Thats why you can attach a small midi controller. Not sure how to go about this.

toneburst, 2019-08-13:
I guess you'd have to select the steps first, then drop into another mode to enter notes with the "keyboard".

The two "blank" buttons S3, S7 could be used to inc/dec the octave.

toneburst, 2019-08-13:
From a hardware perspective, different switch cap colours/shades could be used to mark out the "black" and "white" keys.

I think the screen would have to show something to indicate the buttons were operating in keyboard, rather than step-selection mode.

westlicht, 2019-08-13:
I'm just not too convinced of this idea yet. One option would be to always switch into this keyboard mode when recording is activated. But then there would be the need for yet another option to enable this behaviour, as by default it interferes with existing functionality. It just adds another layer of complexity that I'm not yet sure of its value. The existing methods for entering notes are already plentyful in my opinion, and superior if you don't mind attaching an external keyboard.

dComposer, 2019-08-15:
Yeah this isn't meant to be better than using an external keyboard or anything, just something to sketch an idea with when you don't have an external keyboard handy. Many melodies are within an octave anyway, but I could foresee repurposing the previous/next buttons for a really simple octave shift since I never use these buttons and use the encoder to navigate between parameters usually.

Also, the step buttons aren't used at all for midi tracks, right? This could do double duty there as well for auditioning arpeggiator patterns or if you just want the ability to play a simple melody for a gig but not carry around a keyboard.

ghztomash, 2019-08-28:
Also recording gates by tapping out rhythms on the T1-T8 keys would be pretty nice