westlicht/performer #127: List of USB MIDI controllers that currently work

dComposer, 2019-08-15:
Not an issue, but I thought it would be nice if we had a place to list the USB MIDI controllers that we got to work with the per|former. Feel free to move this to a different spot if you don't want it clogging up your issues.

Here's what I've tested so far:

- Works: CME XKEY37

- Doesn't work: AKAI MPK mini

westlicht, 2019-08-15:
@dComposer Is the AKAI MPK mini the MK1?

I will add a list of compatible devices to the manual. People can add more devices here, I will also do a quick shout out on Facebook to hopefully get some more working devices.

kurtjoy, 2019-08-15:
Off topic..is there a way to directly connect the performer to a laptop via
usb? Desperately trying to get clock out to a laptop that controls video
without using midi interfaces

SloBloLabs, 2019-08-15:
Thats an interesting topic. Any idea how to get yet unsupported devices on the list? I would love to get my Akai mpk mini mkii supported, but for the moment I need to hook a Raspi inbetween to make it work. Libusbhost doesn’t seem to accept the device whereas alink on the other side accepts it natively. I have been in touch with Amid but without the device no way to tackle down with it. I mean, the firmware is all at our hands, so shouldn’t be there a way to solve this? Or would anybody guess these are hardware related incompatibilities on the cortex usb implementation?

SloBloLabs, 2019-08-15:
By the way: Squarp Hermod has same issue. They use F7 as far as I know.

westlicht, 2019-08-15:
@SloBloLabs I would need to have access to an Akai mpk mini to try to figure out why it's not working. From what I've seen the libusbhost implementation is quite minimal and probably does not support the full USB-MIDI specification.

SloBloLabs, 2019-08-15:
Sounds promising! Let’s get in touch how to move on with this. Can I PN you by email?

dComposer, 2019-08-16:
@westlicht it's the Akai MPK mini mk2 ([this guy](https://www.guitarcenter.com/Akai-Professional/MPK-mini-MKII-Limited-Edition-Red.gc)). I'll update it in the original post.

soporus, 2019-08-16:
**Korg MicroKey2** seems to work fine.  Though i haven't tried mapping pitch and mod wheel.  (assuming they aren't mapped by default, because they have no affect by default).   performer fw 0.130

toneburst, 2019-08-16:
Akai APC Mini works as a generic USB MIDI device.

At least the sliders work.

I assume the buttons also send out MIDI note on/off messages (haven't checked with the MIDI monitor), but obviously, the buttons don't light when pressed.

mongrol, 2019-09-01:
Arturia Keystep confirmed not working over USB. It power's on but doesn't appear to be recognised. Performer doesn't report a USB device connected.

sunflowr, 2019-09-04:
Just wondered, does it support USB Hub? Thinking of connecting multiple midi devices e.g. Launchpad and AKAI MPK

smootmather, 2019-10-06:
Arturia MiniBrute and Korg R3 are confirmed as generic USB device.
anybody tried an Arturia MiniLab???

deepiain, 2019-10-06:
Yes - minilab Mk2 working fine here

dcbb, 2019-10-24:
Is it possible to connect an iPad or an iPhone to use a piano roll sequencer app like Xynthesizer?

rheslip, 2019-11-18:
Generic Chinese "USB MIDI" works for me as does the iConnectivity iConnect MIDI

kidko, 2019-12-29:
Keith McMillen QuNexus CV works

acook, 2020-01-20:
Synthstrom Deluge doesn't work via USB. It charges, but no MIDI over USB.

sunflowr, 2020-02-21:
Tested with a 16n faderbank berlind edition:
https://github.com/AtoVproject/16n-AtoVproject-rework

It seems it's not detected when plugged in over USB. I can still get CC values when connecting over the MIDI connector. Don't know if this is an issue with teensy usb-midi library or if it's in performer.

napoleonthegreat94, 2020-06-09:
arturia minilab mk2 is working fine.
arturia microlab is not recognized. is there any way to get it working?! it is my only midi keyboard unfortunately.

rolfjack, 2020-09-25:
I tried hooking up my Novation SL61 mk3 and it registers as a midi device but I cannot register any communication between the keyboard and the performer.

The Novation SL mk3 is designed as a controller to Ableton and has both keys and pads and could potentially be an interesting controller if support was added for it.

Also: I have been looking for some documentation concerning layout of the various Launchpad controllers to get an idea of which controller is the better one to get (and what functionality they each have). If someone has a link to a place where this is described I would be much obliged.

Alexain, 2020-10-10:
Tested Yamaha Modx7 and is recognized. Receive and send midi note and clock.

Tested iPad Pro with camera connection kit and various apps but is not recognized. I think that either have usb-host and can't work. To work with an iPad would need an usb slave (type B connector) on the performer

westlicht, 2020-10-17:
Manual updated.

yagob, 2020-11-05:
Tried Arturia Minibrute 2 and it doesn't work via USB. 
But Akai Max 25 does.

rotadsr, 2020-11-09:
M-Audio Keystation Mini MKI working fine!

CedreLo, 2020-12-08:
AKAI LP25 works well

voltagecontrolled, 2021-01-09:
I have a Deluge and it is working fine here via USB. Performer recognizes it when I plug it in and I'm able to send clock, transport, and notes. I am on version 3.1 of the Deluge firmware. I haven't used it extensively via USB yet so I can't comment on reliability.

notontv, 2021-01-15:
M-Audio Axiom25 works perfectly. Performer finds everything with the "learn" function

SloBloLabs, 2021-03-02:
PR "Fix USB" #291 makes the Akai MPK mini work for me. Further machines may benefit from this patch, as it addresses  low level issues within libusbhost library.

av500, 2021-03-02:
Just a note, all these devices (except the Beatstep Pro) are doing USB/MIDI just fine, all the issues were inside libusbhost. The previously "working" devices were simply the ones that ignored the USB protocol errors of the lib :)