westlicht/performer #128: Octave Transpose for MIDI/CV Track

dComposer, 2019-08-15:
It would be great if the MIDI/CV track had a octave +/- transpose option. I currently use a Beast's Chalkboard between my per|former cv out and my oscillators but it seems like this could be done in the software.

- Now that you've incorporated new midi filtering capabilities, it would be great to be able to drop the lowest note of a chord a few octaves to be used as a bass voice (and vice versa)

- depending on the oscillator I'm using, the arpeggiator tops out while arpeggiating upwards. This way I'd be easily able to knock the arp pattern down a few octaves if I hear this happening in a live setting.

-

westlicht, 2019-08-15:
Will be part of 0.1.31 release.

closed 2019-08-15 by westlicht

westlicht, 2019-08-16:
Oh just realized yoy where asking for octave and not transpose parameter.

commit 915c66fb (2019-08-15): added transpose to midi/cv track

fixes #128