westlicht/performer #130: Launchpad suggestions

sunflowr, 2019-09-03:
When editing note values on the launchpad. It would be nice if the buttons labeled 1-8 above would light on the steps with gate on so I can easily see what steps will play.
It would also be nice if A-H would light up showing what vertical position in the window I'm currently at (what octave I'm currently at).

A Ableton Push style layout would be sweet as well when editing ;)
It would be cool with the note layout at least for selecting pitch.

westlicht, 2019-10-07:
The current layout uses the extra row/column for other functions, so no easy way to do that. I did already think about alternate layouts similar to Push but it's not high priority.

Volod3Mir, 2019-11-18:
It would also be great if the launch pad would follow/ sync to page changes on the performer and vice versa, as of now it seems that it's one of the few only places where it doesn't. 

Also it would be nice to have some control implemented over randomization. Top row buttons 6 and 7 are yet unused, and it would be a perfect addition to quickly randomize the tracks.

Thanks,
Cheers!

Materia-Oscura, 2020-08-25:
+1 for the Follow/Sync page idea from Volod3Mir!