westlicht/performer #131: Sometimes a track does not start to play

froschmett, 2019-09-09:
Hi,
I sometimes experience that a single track does not start to play. I do not know which action leads to this behaviour, because I did not keep track of those actions before it happens. But I worked a lot on a track also made an init and I think it happened once direclty after an init.

I have to restart the performer and then it works again. 

Maybe you can make some sense out of this. Great sequencer nonetheless!! :)

westlicht, 2019-10-07:
Cannot reproduce this behaviour, never observed it before. If you find a way to reproduce that would be awesome.

froschmett, 2019-10-07:
Hey,
I will keep an eye on this. It happened twice so far on latest fw. Haven't had it since...

patrickod, 2019-10-08:
Just adding a 👍 as I've also observed this bug a handful of times, but have not yet found a reproducible case.

froschmett, 2019-10-21:
Happend again, cant tell, what exactly lead to it, but I fiddled around with T1 to T3. First, I configured T1 as Midi to CV to test "virtual MIDI", then after some playing around I switched T1 to Curve mode... Then I select all 16 steps at once and turned the encoder for several track settings like min, max and shape... After hitting play T2 and T3 played, T1 did not start. I saved the project, switched the Performer off and on, then everything worked again... Strange indeed...

gitmusik, 2019-11-20:
This happened to me (twice in one session).

It was after I converted a track from note to curve.
I edited the curves quite extensively.
And then initialised the track.

After two more similar transformations (curve to note and note to curve) the tracks began playing again.

HOWEVER...I just tried that again and I couldn’t repeat the weird behaviour.

westlicht, 2019-12-01:
@gitmusik did this happen when the track was in curve mode or in note mode? I mean when it not played.

gitmusik, 2019-12-01:
In my case, the track would not play when I was in curve mode.

napoleonthegreat94, 2019-12-26:
Hello i've encountered this problem too. It occured to me when changing a note track to a curve track

westlicht, 2020-01-17:
Finally managed to reproduce this bug. Fix is coming in 0.1.34!

commit a1300154 (2020-01-17): lock engine when changing track modes

fixes #131
closed 2020-01-17 by westlicht