westlicht/performer #134: PerFormer Restart if setting a Tracklayout to MIDI/CV

smootmather, 2019-09-20:
Hi Simon,

first thanx for your incredible work. Performer is such a great Seq. Interactive BOM and Build guide made it an easy "gothrough". the
Right now i´m trying to explore midifunctions. 
my mididevice is an Arturia MiniBrute or an Korg R3. both working both ways. MidiJack und USB midi. 
midi is received at the moment when i´m connecting the device no need to set a track to MIDI/CV. i can record midi notes to track which is selected.
but if a track is set to MIDI/CV and PerFormer is receiving a MIDI signal the module reboots. 
i tried both midi options. USB and MidiJack. both devices same behavior.
did i miss something or is this NOT the way it should work.
Kind regards. Manuel

westlicht, 2019-09-21:
Does this happen with both the R3 and minibrute?

smootmather, 2019-09-21:
Yes with both. On both ways USB and midijack. 

---- Simon Kallweit schrieb ----

westlicht, 2019-09-21:
With newest firmware and without loading a project first?

smootmather, 2019-09-21:
Yepp. Latest Firmware. No project loaded. 

---- Simon Kallweit schrieb ----

westlicht, 2019-09-21:
That is rather strange. I will try to reproduce. But if you can give the extract instructions of how to reproduce this, I can probably reproduce it on my end.

smootmather, 2019-09-21:
I'll make a video tomorrow with exact instruction and behavior. 

---- Simon Kallweit schrieb ----

smootmather, 2019-09-22:
Hi Simon, hope this helps. Midi is coming from MiniBrute. Same behavior with Korg R3. Selected track receives Midi and sends it to dedicated output (CV and Gate). Track is in Note Layout. When track is set to Midi/CV Layout and Midi Source is selected the module resets. 

https://youtu.be/YZ0zmu0Djco

https://youtu.be/HDBm29e2tt4

---- Simon Kallweit schrieb ----

westlicht, 2019-09-22:
Thanks for the detailed bug report. Will try to reproduce this evening.

westlicht, 2019-09-22:
@smootmather I just tried to reproduce the bug but had no luck. I actually have a MiniBrute too so the test settings were pretty much equal. Can you retry without the SDCard in the slot (also on startup) to make sure you have the default init settings. If the reset does not happen with the SDCard out, it's related to something stored in your patch. Then please attach that patch by copying it from the SDCard.

westlicht, 2019-09-22:
@smootmather also you are running the latest 0.1.31 release, right?

smootmather, 2019-09-22:
Restarted it without SD Card. Same behavior. Could it be an Hardwareproblem. something soldered not probably...should i try to reflash the stm? 

---- Simon Kallweit schrieb ----

westlicht, 2019-09-22:
Because it happens both with USB and MIDI I didn't think it's a hardware issue, at least not an obvious one. I'm a bit in the dark here unfortunately.

smootmather, 2019-09-24:
Hey Simon, i reprogrammed the STM and everything works as it should. Sorry for wasting your time. Thanx again for this incredible sequencer.

westlicht, 2019-09-24:
Glad to hear that!

closed 2019-09-24 by westlicht