westlicht/performer #135: Sync performer to a multi-track tape.  SMPTE, Midi sync + SPP, MTC, MMC?

guytaylor, 2019-09-22:
Hi,
I wanted to be able to sync a multi track analog tape to the performer.
Two options sprung to mind:
1. (The slicker one). Let the performer stripe SMPTE and be able to read it via one of the CV inputs and control the song position that way
2. (Bit more clunky) Let the performer read/write Midi sync, SPP and/or MTC and use an external hardware box to connect to the tape (like a JL Cooper PPS 100)

Another cool feature could be to implement MMC, that way the two devices could control each other :)

westlicht, 2019-10-07:
I don't really see the point of the song position message. Sure, I could send SPP=0 when resetting the clock but I thought that was what the start message would imply, because there is also a continue message. Sending SPP messages as the sequencer is running makes no sense as there is never a jump in song position.

Ideas 1/2 seem way out of what performer is intended to be. I assume there are devices that either take MIDI clock and generate MTC or vice versa, so that might be a much better option to sync up these different devices.

guytaylor, 2019-10-08:
Hi,
Sorry I was a bit unclear!

I just meant:  

“Could the performer RESPOND to SPP messages. “

That way song mode could skip to the correct position in a song when receiving an SSP midi message

As SPP messages are just a number of ‘ticks from the start of the song’, I assume on receiving one you would you would calc which part of the song should be started at and start playing from there.

This would be a great feature as it means I could lock the performer to my external tape/sequencer.

And yes, forget 1/2 option. Way too complicated and I can do it with an SMPTE to midi converter :)

Thanks
Guy

westlicht, 2019-10-08:
I see, well, this is a bit tricky to implement in the way song mode is currently implemented. But I might give it a thought in the future.