westlicht/performer #136: Patterns and MIDI Program changes

guytaylor, 2019-09-25:
Hi,
I was hoping that the following could be added:

1. The ability to change to a pattern via a MIDI program change
2. The abilty for a program change (or set of them) and/or other MIDI message to be sent when a changing a pattern 

Thanks

westlicht, 2019-10-07:
With the current concept of how to generate MIDI output data this is a bit tricky. Reacting to program changes might be feasible though.

kidko, 2019-12-26:
I second the request for the ability to send program changes to MIDI out. I was hoping to control my Digitakt patterns with Performer

cnicht, 2020-03-14:
Yes it would be a great idea!

greengrooves, 2021-09-02:
Would love a feature like that also for sending the program change out to the cv bus

jackpf, 2022-08-07:
I've implemented sending program changes (not receiving though, yet) here in case anyone finds it useful: https://github.com/jackpf/performer/pull/10