westlicht/performer #138: Step repeat

madeinspace, 2019-10-02:
Any chance of implementing a step repeat? that'd take us closer to #105 :) thanks for considering!

deepiain, 2019-10-02:
Already has it?

madeinspace, 2019-10-02:
> Already has it?
> […](#)

I don't think so, I'm not talking about rachets, but step repeat per pulse as in the possibility of repeating the step for each pulse as opposed to rachets

westlicht, 2019-10-07:
Step repeat is not implemented and will not become part of note sequence. This unfortunately uses up too much memory and I would have to reduce the number of patterns of a project which I would like to avoid. But there are plans to have this functionality in a different track mode as #105 mentions.

jstefani, 2019-10-07:
I have been thinking of a version a bit like the beat roller (I think that is the name) on the BeatStep.

Would work as follows:

Perf page:

hold down s1 to repeat last 1 step, s2 to repeat last 2 steps, etc. or something similar

Another option:

s1 = repeat last 1 of track 1
s9 = repeat last 4 of track 1 

s2 = repeat last 1 of track 2
s10 = repeat last 4 of track 2

etc.

perhaps we make the pre-set repeat number settable by the user

Will give it a try after I've looked at the code more. Hopefully there is some global pointer to each tracks current step.

westlicht, 2019-10-08:
All the step keys are already used on the performer page. Holding `S1-8` allows for setting the fill amount. Pressing `S9-10` activates fills.

jstefani, 2019-10-09:
ah, nice. missed that. perhaps F5 enables the beat roller/looper. speaking of course a feature that doesn't yet exist and perhaps no one wants :)

westlicht, 2019-10-18:
I thought about adding the beat repeat thing as a fill mode. But having it set to a fixed repetition value is not that cool, so it would still require additional UI which is kind of hard at this point.

closed 2020-01-17 by westlicht