westlicht/performer #139: Latched Pattern change does not work with selected track

smootmather, 2019-10-06:
Hey Simon,

latched pattern change does not work for any selected track. any other channel reponses to the latch function. tried it with any channel.
latest firmware is installed

kind regards. manuel

westlicht, 2019-10-07:
I cannot reproduce what you describe. Here is what I tested.

- created two patterns on track 1
- play pattern 1
- press `PATTERN` + `F1` + `S2` to switch to pattern 2 in latch mode
- release buttons

I also tried the other way to use latched pattern change:

- press `PATTERN` + `F1` but then keep holding `F1` while releasing `PATTERN` and then press `S2` to switch to pattern 2 in latched mode

I even tried only doing it for track 1 in isolation using button `PATTERN` + `F1` + `T1` + `S2` or the second form described above.

All work as expected.

smootmather, 2019-10-08:
that`s the way it works on mine too. but when going to permanent pattern page  and try to latch isolated pattern which  is selected e.g. in step page, latching for this track is not working. same when trying to latch every track in permanent pattern page.
Hope this describes it better. otherwise i can make a video.

westlicht, 2019-10-08:
Perfect, should be able to reproduce it now and fix it! Thanks for reporting.

westlicht, 2019-10-08:
Will be part of 0.1.32 release.

commit 4bb4f79d (2019-10-08): fixed latch and sync mode on pattern page

latch/sync was ignored due to page re-enteriing. fixes #139
closed 2019-10-08 by westlicht