westlicht/performer #143: feature request: arps for cv tracks

froschmett, 2019-10-11:
Hi,

wouldn't it be possible to make arps available on cv tracks? Like defining a root note, chord type and arp mode? Happy to hear some feedback :).

toneburst, 2019-10-18:
There's already an arpeggiator for CV tracks.
Please look in the [manual](https://westlicht.github.io/performer/manual/) before submitting a feature-request.

westlicht, 2019-10-18:
@froschmett did you mean on normal note tracks? In that case no, there is no plan to support that. As @toneburst mentioned there is a pretty powerful arp available on MIDI/CV tracks.

froschmett, 2019-10-18:
Hey,
I read the manual, thanks for mentioning... What if I don't use MIDI to CV?

Anyway, I am using the 2hp arp module to achieve the effect I need.

westlicht, 2019-10-18:
Ok, that's great then. MIDI/CV is meant for studio use and to be able to play some voices live, where the arp makes a lot of sense. You can configure the sequencer to take in CV/Gate on the CV inputs and use that as "virtual MIDI" on a MIDI/CV track, so essentially you'd have the option to run this from some other source than a MIDI keyboard.

closed 2019-10-18 by westlicht

froschmett, 2019-10-19:
Hey,
ok, I will test that, thanks Westlicht :)!

froschmett, 2019-10-21:
Hmm,

tried it, cant get that to work, maybe you can provide an example? Would be nice, thanks a lot... :)