westlicht/performer #144: double tap SHIFT too fast?

mopoco-git, 2019-10-14:
Hi there,
i'm new to the performer and absolutely loving it so far!

on the STEPS page: i find the double tap SHIFT function to select all steps quite hard
to achieve. 
could (and should?) the recognition of the double tap be set so that we can double 
tap with longer intervalls? (it generally makes me a bit nervous, especially when i'm working on slower sequences...)

froschmett, 2019-10-14:
2nd that, you got to be real fast to achieve selections with 2x shift...

westlicht, 2019-10-18:
Ok, will experiment with slower double taps, i got kind of used to the fast ones :)

mopoco-git, 2019-10-20:
awesome! and since afaik there's no single-shift-function (only hold-shift functions) i guess you
could make the double tap recognition basically reeeeeeally slow and your fast ones will still get recognized? though an intervall of one second might be enough, even for the most chilled out amongst us  :)

westlicht, 2019-11-11:
Double tap interval will be increased by 50% in 0.1.33 release.

commit 9a129248 (2019-11-11): increased double tap time from 200ms to 300ms

fixes #144
closed 2019-11-11 by westlicht