westlicht/performer #145: New Novation Launchpad Models Support

toneburst, 2019-10-16:
Novation have just released two new Launchpad models, the Mini Mk3 and Launchpad X.

Can we have support for these, please?

I have a Mini Mk3 on preorder at the moment. I’m happy to help test support, when it arrives (hopefully in a couple of weeks).

It would be really great to have support for the RGB pad lights. 

Something like a different set of colours for each of the 8 tracks would be great!

toneburst, 2019-10-16:
The "X" version has velocity and pressure (channel aftertouch, I guess). I wonder if this could be utilised in some way when editing.

Just throwing that out there...

toneburst, 2019-10-18:
My Mini Mk3 has been delivered :)

westlicht, 2019-10-18:
Tell me the vid/pid once it arrives so I can add it to the list. Color support might follow.

toneburst, 2019-10-18:
I tried plugging it in to the Per|former and it powered on, but was stuck in “screensaver” mode, and didn’t show any output in the module’s MIDI monitor when pressing buttons.

I will investigate further in a little while, and report back my findings.

toneburst, 2019-10-18:
Page 22 of this document has the default mappings for “Programmer” mode.

https://customer.novationmusic.com/sites/customer/files/downloads/Launchpad%20Mini%20-%20User%20Guide.pdf

I think it may have to be switched to this mode to work with Per|former.

Will test and confirm.

westlicht, 2019-10-18:
Was just trying to find the "Programmers Reference Guide" as suggested in the manual, but couldn't find a download. But I assume the vid/pid is still a new one. So the sequencer will not detect it as a launchpad controller. You can add the vid/pid yourself if you know how to compile the firmware for testing. Just add it here: https://github.com/westlicht/performer/blob/master/src/apps/sequencer/ui/ControllerManager.cpp#L4

toneburst, 2019-10-18:
MIDI notes are received from the LP Mini Mk3 when it’s in Programmer mode.

Annoyingly, it won’t start up in this mode. You have to manually switch to Programmer mode every time it starts up.

Will try and get the ID for the device now (if I can remember how to do it).

toneburst, 2019-10-18:
Device info, from MacOS System Report:

Product ID: 0x0113
Vendor ID: 0x1235
Manufacturer: Focusrite - Novation

westlicht, 2019-10-18:
Yep, so that is not currently in the list. I can quickly send you a build for testing if you cannot compile yourself.

toneburst, 2019-10-18:
There doesn't seem to be any info about how to set pad colours. I assume its done via Velocity data attached to the note events sent to it, in the same way as with the LP Pro and full-size Launchpad.

Don't know if it uses the same palette of 128 colours.

westlicht, 2019-10-18:
Yeah, I was just wondering if it uses the same protocol as the Pro because it has colour leds. So you'd also have to adjust this line: https://github.com/westlicht/performer/blob/master/src/apps/sequencer/ui/controllers/launchpad/LaunchpadController.cpp#L102

toneburst, 2019-10-18:
> I can quickly send you a build for testing if you cannot compile yourself.

That would be great!

toneburst, 2019-10-18:
You also support the full-size LP Mk.2, too, don't you?
https://novationmusic.com/launch/launchpad-mk2
Might be more like this, as the pads don't have velocity-sensitivity.

westlicht, 2019-10-18:
Here is a build with the new vid/pid using the LP Pro protocol.

[performer-launchpad-mk3.zip](https://github.com/westlicht/performer/files/3745433/performer-launchpad-mk3.zip)

toneburst, 2019-10-18:
Cool. Will try that now.

toneburst, 2019-10-18:
OK, well, it *sort of* works...

The module seems to receive the expected data when pads are pressed, but none of the pads light up.

I doubt the note numbers are incorrect, as I'd expect something to light up, even if not the expected pads, if the mapping is wrong.

Maybe the the Mini Mk3 expects to receive notes on a different channel to the Pro.

westlicht, 2019-10-18:
While at it, can you also try this version which uses the Mk2 protocol?

[performer-launchpad-mk3-take2.zip](https://github.com/westlicht/performer/files/3745487/performer-launchpad-mk3-take2.zip)

toneburst, 2019-10-18:
Pads start on note #11 on the bottom-left pad and go up in increments of 10 for each row.

etc.
21 22 23 24 25 26 27 28 29
11 12 13 14 15 16 17 18 19

toneburst, 2019-10-18:
> While at it, can you also try this version which uses the Mk2 protocol?

Will try that now.

toneburst, 2019-10-18:
> While at it, can you also try this version which uses the Mk2 protocol?

Same deal. Pads seem to work (though it's hard to tell), but don't light.

westlicht, 2019-10-18:
Ok, in that case we have to wait for the programmers guide to appear. I didn't find it yet, but please let me know when you find that document.

westlicht, 2019-10-18:
I think I'll order one of these new LP as well for testing.

toneburst, 2019-10-18:
Update: the matrix pads work, but the buttons around the sides don't.

toneburst, 2019-10-18:
> Ok, in that case we have to wait for the programmers guide to appear. I didn't find it yet, but please let me know when you find that document.

I'll keep my eye out for it.

toneburst, 2019-10-18:
OK, I've learnt some more.

The 64-pad matrix sends and responds to NOTES

`
81 82 83 84 85 86 87 88
71 72 73 74 75 76 77 78
61 62 63 64 65 66 67 68
51 52 53 54 55 56 57 58
41 42 43 44 45 46 47 48
31 32 33 34 35 36 37 38
21 22 23 24 25 26 27 28
11 12 13 14 15 16 17 18
`
The pads around at top and right send and respond to MIDI CCs

`
91 92 93 94 95 96 97 98 (99 LOGO)
                        89
                        79
                        69
                        59
                        49
                        39
                        29
                        19
`

toneburst, 2019-10-18:
I'm thinking the separate devices might actually be useful, if they can both be accessed.

If the Per|former treated one of virtual devices as a generic class-compliant MIDI device, and the other as a pad controller, it would provide a way of entering note data in a more conventional way, using the LPs builtin _Keys_ mode.

Switching the LP back to _Programmer_ mode, you could then use the LP as a pad controller again.

toneburst, 2019-10-18:
I think my long post above should have most of the info needed to make it work.

Let me know if you need me to run any more tests, though.

I think I'll be buying another one of these things!

toneburst, 2019-11-08:
Novation have released the Programmer's Reference manual for the LP Mini Mk.3.

https://www.dropbox.com/s/6e1uyolmxocms3r/Launchpad%20Mini%20-%20Programmers%20Reference%20Manual.pdf?dl=0

westlicht, 2019-11-08:
Awesome, will add support soon.

westlicht, 2019-11-11:
Quick update on the Launchpad Mk3/X support:

Receiving button presses works fine, but the new launchpad uses yet another mapping. This was the easy part. The harder part is that sending the LED state is now done on a second MIDI _cable_. Multiple MIDI _cables_ per USB MIDI device are not currently supported. I quickly tried hacking the driver to allow for that but I get a ton of NAKs when sending LED states. Some messages are sent over successfully, but many of them are dropped. I don't yet know what the actual problem is, but I will investigate further.

toneburst, 2019-11-12:
I think it’s worth cracking, because I suspect Novation are going to use a similar dual-interface setup in future LP (and maybe also LaunchKey) models.

The new full-size Launchpad X uses the same system, according to the Programmers Reference doc

https://customer.novationmusic.com/sites/customer/files/downloads/Launchpad%20X%20-%20Programmers%20Reference%20Manual.pdf

westlicht, 2019-11-12:
Device descriptor dump for Launchpad Mini Mk3:

```
Bus 002 Device 011: ID 1235:0113 Focusrite-Novation 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.10
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1235 Focusrite-Novation
  idProduct          0x0113 
  bcdDevice            2.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 3 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x007e
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          4 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           0
      bInterfaceClass         1 Audio
      bInterfaceSubClass      1 Control Device
      bInterfaceProtocol      0 
      iInterface              5 
      AudioControl Interface Descriptor:
        bLength                 9
        bDescriptorType        36
        bDescriptorSubtype      1 (HEADER)
        bcdADC               1.00
        wTotalLength       0x0009
        bInCollection           1
        baInterfaceNr(0)        1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         1 Audio
      bInterfaceSubClass      3 MIDI Streaming
      bInterfaceProtocol      0 
      iInterface              6 
      MIDIStreaming Interface Descriptor:
        bLength                 7
        bDescriptorType        36
        bDescriptorSubtype      1 (HEADER)
        bcdADC               1.00
        wTotalLength       0x0043
      MIDIStreaming Interface Descriptor:
        bLength                 6
        bDescriptorType        36
        bDescriptorSubtype      2 (MIDI_IN_JACK)
        bJackType               1 Embedded
        bJackID                 1
        iJack                   7 
      MIDIStreaming Interface Descriptor:
        bLength                 9
        bDescriptorType        36
        bDescriptorSubtype      3 (MIDI_OUT_JACK)
        bJackType               1 Embedded
        bJackID                 2
        bNrInputPins            1
        baSourceID( 0)          1
        BaSourcePin( 0)         1
        iJack                   8 
      MIDIStreaming Interface Descriptor:
        bLength                 6
        bDescriptorType        36
        bDescriptorSubtype      2 (MIDI_IN_JACK)
        bJackType               1 Embedded
        bJackID                 3
        iJack                   9 
      MIDIStreaming Interface Descriptor:
        bLength                 9
        bDescriptorType        36
        bDescriptorSubtype      3 (MIDI_OUT_JACK)
        bJackType               1 Embedded
        bJackID                 4
        bNrInputPins            1
        baSourceID( 0)          3
        BaSourcePin( 0)         1
        iJack                  10 
      Endpoint Descriptor:
        bLength                 9
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
        bRefresh                0
        bSynchAddress           0
        MIDIStreaming Endpoint Descriptor:
          bLength                 6
          bDescriptorType        37
          bDescriptorSubtype      1 (GENERAL)
          bNumEmbMIDIJack         2
          baAssocJackID( 0)       1
          baAssocJackID( 1)       3
      Endpoint Descriptor:
        bLength                 9
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
        bRefresh                0
        bSynchAddress           0
        MIDIStreaming Endpoint Descriptor:
          bLength                 6
          bDescriptorType        37
          bDescriptorSubtype      1 (GENERAL)
          bNumEmbMIDIJack         2
          baAssocJackID( 0)       2
          baAssocJackID( 1)       4
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         8 Mass Storage
      bInterfaceSubClass      6 SCSI
      bInterfaceProtocol     80 Bulk-Only
      iInterface             11 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
```

westlicht, 2019-11-12:
Got the new launchpad up and running after fixing a few things in `libusbhost`. However, for some reason sending MIDI messages to the second port for controlling the lights still results in many NAKs. I only fixed this by resending messages that have been NAKed, but this results in quite slow transmission due to the polling nature of `libusbhost`, so while this is useable it's far from optimal. I don't yet know why the NAKs are happening. I assume they originate from the launchpad, but this seems rather weird. Also, I still have to implement sending the SYSEX message for enabling programmers mode when plugging the device in.

toneburst, 2019-11-12:
Wild stab in the dark, but is anything still being sent to the DAW-control port?

Maybe it’s receiving the same messages simultaneously on both ports/interfaces, and that’s what’s confusing it.

westlicht, 2019-11-12:
Nope, I actually hacked the driver to always send to the second port :) This obviously breaks with other MIDI devices but I'm really just trying to get the transmission fixed right now.

toneburst, 2020-02-02:
Any progress, or further thoughts on this?

Still happy to help with testing etc. My Launchpad Mini Mk.3 is gathering dust at the moment.

toneburst, 2020-05-04:
Any further progress on this?

westlicht, 2020-06-14:
@toneburst sorry, didn't make any progress on this yet. It's not high priority as all previous launchpads work fine and this requires quite some more time to fix which I currently don't have.

Pipstapee, 2020-07-08:
This may have been covered, Is the Pro Mk 3 compatible?

Waldemar-Johann, 2020-09-13:
Any progress to this. I'm eyeing on a Pro  MK3 too.

Thanks for answer.

westlicht, 2020-10-24:
Good news, just updated my launchpad mini mk3 with the new firmware and it behaves much nicer. I still need to add proper support for multiple _virtual cables_ because the new launchpad operates on two cables, one for daw, the other for controlling the pads in live/programmer mode. I also need to add support for sending SYSEX so programmer mode can be initiated automatically on startup. But it seems much more in reach now!

toneburst, 2020-10-25:
That’s cool! Good thing I didn’t get round to selling my Mini Mk.3.

westlicht, 2020-10-26:
Support will be released with 0.1.39

closed 2020-10-26 by westlicht