westlicht/performer #146: Midi issue

Kooza25, 2019-10-16:
I’m trying to connect a usb midi interface with an ipad. Midi out from ipad is working with other gear, but performer doesn’t receive signals, monitor page is empty.
Same thing when trying to have an output, track has correct settings but no signal.
Is happening to another guy who’s running 2 performers.
Running latest update

kurtjoy, 2019-10-16:
Same problem here

westlicht, 2019-10-18:
How exactly do you connect to the ipad? Directly via usb will not work as both are usb hosts. If you use a usb to midi adapter on the ipad make sure you use the correct midi trs type a adapters to connect midi to the performer.

Kooza25, 2019-10-18:
Of course is a usb-midi interface which is eorking correctly with any other outboard and I’m using working midi-tra cable

westlicht, 2019-10-18:
Ok, so the same midi trs cable works using a different (or same) midi interface not connected to the ipad? If so, thats a bit weird, if not either the cable is defect (or the wrong type) or the midi input on your performer is defect.

Kooza25, 2019-10-18:
I have two midi-trs, both non working with performer. Problem is nor the midi out works. I checked all the midi related components, no shorts.

SloBloLabs, 2019-10-18:
Can you check the type of cable you are using? There are two types of trs midi standards out there. I use the make noise adapters which are properly working with the sequencer.

Kooza25, 2019-10-18:
One of em is the one included in lil erebus kit.
Any advice on how to check it?

SloBloLabs, 2019-10-18:
Sure, you can check midi.org for further details:

https://www.midi.org/articles-old/updated-how-to-make-your-own-3-5mm-mini-stereo-trs-to-midi-5-pin-din-cables

Kooza25, 2019-10-18:
So, performer requires midi a or b?

SloBloLabs, 2019-10-18:
Performer is standard A.

Kooza25, 2019-10-18:
Yes I confirm mine are type b. I’m converting one and see if it works, you’ve been very helpful!

SloBloLabs, 2019-10-18:
Welcome. Let us know if it works to close the issue.

westlicht, 2019-10-18:
> Yes I confirm mine are type b.

Yes, that would explain why it's not working for you. FYI, unfortunately the MIDI association was a bit late in the game on standardizing the MIDI TRS plug. But Performer is using the official standard (type A) to be future proof. Unfortunately there are a lot of type B adapters in circulation by now.

kurtjoy, 2019-10-18:
Hey Simon. My issue is slightly different. I've tried multiple interfaces,
made and bought both A/B trs to midi, with no luck. I've tested the
interfaces and cables with other gear and they work fine.

Tried a variety of software (midiox, reaper, hollyhock) to receive and send
midi, with no luck.

Could it be a problem with the build?

westlicht, 2019-10-18:
@kurtjoy yes, that definitely sounds like a hardware issue. You probably have to do some hardware debugging or just start by reflowing the relevant components on the MIDI in/out circuit.

SloBloLabs, 2019-10-18:
Another issue with the midi jacks I encountered is you should not solder them planar onto the pcb. Let them stand up a bit more out of the front panel so the screw goes a bit further down the thread. Otherwise - as the stereo jacks are a bit shorter than the others - the pin of the plug does not get down into the jacks for proper connectivity.  Plug you adapter in and use a multimeter to check if all three lines are joint or if this relates to your issue.

westlicht, 2019-10-18:
> Another issue with the midi jacks I encountered is you should not solder them planar onto the pcb. Let them stand up a bit more out of the front panel so the screw goes a bit further down the [...]

Yep this is important too and documented in the build guide if I remember correctly :)

kurtjoy, 2019-10-18:
I bought the unit second hand. So will check on both of those things. Thanks

Kooza25, 2019-10-18:
Yes, issue solved for me. Many thanks!

closed 2019-10-18 by westlicht

antoniograzioli, 2019-10-18:
same issue here, I was discussing it with @Kooza25 last night...we wre eusing the same MID-TRS cables (in my case the ones included in the BeatStep Pro package). I will try to build some suitable cables, then I am sure I won't have any issues...

SloBloLabs, 2019-10-19:
Indeed, the BSP adapters are exactly the ones that don’t work. These are type B.

noahsider, 2020-07-06:
Hi, I have this same issue (midi in and out don't work at all), even with a type A cable. The midi jacks read as connected correctly from my multimeter so I'm not sure what the next step should be.