westlicht/performer #148: MIDI output issues

mongrol, 2019-10-27:
v1.32: Midi output only sends data when the note's are C#4 and upwards. Anything from C4 (0v) or below results in nothing being sent.
In addition midi-in to midi-out doesn't pass through no matter what the note or whether Play mode is activated or not unlike midi-in to CV out which works fine.

mongrol, 2019-10-27:
Further testing reveals it's a velocity issue. The note data is being sent but with velocity of zero.

westlicht, 2019-11-11:
> In addition midi-in to midi-out doesn't pass through no matter what the note or whether Play mode is activated or not unlike midi-in to CV out which works fine.

What do you mean exactly?

rheslip, 2019-11-18:
Related enhancement would be to pass MIDI thru when step recording for audible feedback when a MIDI synth is attached.

westlicht, 2020-01-18:
Implemented monitoring forwarding to midi output. Will be released with 0.1.34.

commit a628fa92 (2020-01-18): send midi output on monitoring

fixes #148
closed 2020-01-18 by westlicht