westlicht/performer #153: MIDI import tool?

dComposer, 2019-11-10:
Not an issue, but would you have any interest in writing a helper MIDI import tool that converts MIDI files into projects you can save to the SD card? I only ask because I have tons of drum MIDIs I’d love to play with on the performer but it would be annoying to manually key them in. I tried peeking into a project to see if I could figure out the code and ended up corrupting the project I was messing with 😆

westlicht, 2019-11-11:
I have previously thought about a converter utility to convert projects to json and back. This would make it fairly straight forward to do these kinds of things.

patrickod, 2019-11-11:
@westlicht is there any existing documentation on what you think the specification of this JSON format would look like? Wouldn't mind taking a shot at a conversion utility potentially

westlicht, 2019-11-11:
@patrickod it would basically just be a representation of the model class hierarchy starting with the _Project_ class as a root. I would first write a JSON schema probably. Are you interested in working on this or did you refer to the utility that would take MIDI and output a JSON file?

dComposer, 2019-11-11:
I was planning to make a simple python script to convert the midi once I got my hands on that json file. Of course I’ll share whatever I come up with!

westlicht, 2019-11-20:
Actually I think making a python library will make more sense as i already have bindings for much of the data model. But its some work that i currently dont have time for.

dcbb, 2020-01-27:
+1 for MIDI import!

I'm happy to join or support developing that converter if it's Python based.

toneburst, 2020-02-02:
+1 for MIDI-to-Project converter/script.

dComposer, 2020-05-07:
I just thought of something that would be really cool -- what if I could store MIDI files on the SD card and load them straight into a project through the UI (instead of having to convert them to the performer project format with a script on my computer first)? That would be incredibly handy with a DAW to hardware workflow!

jasonnolan, 2020-08-07:
I support any of this sort of thing. Presently, I use my Disting MK4 for playing midi files, but it is not the Performer. Happy to throw a gift card at anyone taking it up. All that wonderful performer real-estate just waiting for my midi files. I think MIDI-to-Project converter/script would be most useful, if it would result in 1-8 midi files being incorporated for 1-8 gates and CVs outputted.

Materia-Oscura, 2020-08-29:
+1 for a midi to project convertor
+1000 for a midi compatible module than allows to load midi file staight form the sd card without converting it !!!

polymer81, 2021-07-28:
This is the only thing I need more from my performer. There are some available ones, maybe they can be of use ? 

https://tonejs.github.io/Midi/
https://github.com/Tonejs/Midi

Thanks for an amazing sequencer !