westlicht/performer #154: Start Song from External Midi device

DF100, 2019-11-11:
Per|former needs to receive start/stop message from external midi devices during/while in song mode.
Song mode sequences to begin only when midi start message recieved.

westlicht, 2019-11-12:
Ok, I just played around with this a bit. There is already a pretty good way of doing this:

- set _Mode_ on clock page to _Slave_
- on the song page you can now select the row you want to start from and press _Play_ (`F5`)
- start the external MIDI clock

This will start the song playback in sync with the external clock. I think this is what you want?

It's important to understand that song mode is not intended to turn the sequencer into a linear playback device. You can do that by simply starting from the beginning and not interacting. But the design was done in a way that allows to engage/disengage song mode while the sequencer is running.

DF100, 2019-11-13:
Hi Simon, thanks, I already tried that and the problem is the sequencer is running already. 
My octatrack is the master device in my live setup and follows an arrangement. I need the performers to follow the same arrangement. So i need to load the song in song mode 'ready to go' but not actually do anthing until I hit go on the octatrack. Technically 'waiting' in the starting blocks if you like. Then everything plays when its supposed to and I can concentrate on the sounds rather than the patterns played.

Best Regards

Dave

westlicht, 2019-11-13:
If the sequencer is in slave mode it is only running when your octatrack is running. That is the important bit. If it is in auto mode, you cannot "arm" the sequencer without starting it's internal clock. When its in slave mode and no midi clock is coming in, it is "waiting" in the starting blocks as you said. Otherwise please elaborate more on what exactly you try to accomplish.

DF100, 2019-11-13:
No thats exactly what i want. Except I tried that and it didnt behave any different than in 'Auto' is all. 
Will try it again tonight.

Best Regards

Dave

DF100, 2019-11-16:
So I managed a work around I turned the two performers into 16 midi to
CV/Gate converters and am sequencing straight out of Ableton Live. Works a
treat Thanks.

westlicht, 2019-11-16:
Sure, you can use performer as a midi/cv interface :) it's not its main purpose, but whatever works. I will close this as I think there is a solution to the problem using _Slave_ clock mode.

closed 2019-11-16 by westlicht