westlicht/performer #156: No midi lower than D#0

smootmather, 2019-11-12:
Hi Simon, not sure if it depends on the update. but there´s no midi send with notes lower then D#0

regards manuel

westlicht, 2020-01-16:
Not sure what you mean but I can easily output the full range.

- C0 on the sequencer (0 volts) represents MIDI C3 (decimal 60)
- The lowest representable note in MIDI is C-2 (decimal 0) which corresponds to C-5 (-5 volts) on the sequencer.
- The highest representable note in MIDI is G8 (decimal 127) which corresponds to G5 (> 5 volts) on the sequencer. The highest note that can be output physically on the sequencer is C8 (5 volts) but internally the sequencer can generate notes greater than that, if a suitable scale is selected.

I've just tested this and all seems fine, note sure how you came up with D#0 as the lowest note.

closed 2020-01-16 by westlicht

smootmather, 2020-01-16:
Hello Simon, 
this is a videolink which describes my problem. https://youtu.be/pnCLEVi06qQ
Midi is sent via USB to MiniBrute. Problem also occurs  when midi is send via midi. If i take another midisource, lower notes work on the MiniBrute. 
Hope that helps. Maybe something is wrong with my Device, but another one i built for a friend had the same problem... 
Kind Regards. Manuel 

---- Simon Kallweit schrieb ----

westlicht, 2020-01-16:
Really weird. I tested with a midi monitoring application to inspect the messages actually sent on the wire and it's all good. What's your midi output setting on performer?

smootmather, 2020-01-16:
I also tried with another Synth to receive midi. 

---- Simon Kallweit schrieb ----

smootmather, 2020-02-01:
Hi Simon,  today i found that i have to set velocity to min. 1 that midi is send C0. If velocity is set to track than only notes  above C0 is getting out. 
Hope that's helps. 
Regards Manuel