westlicht/performer #161: Fill unmute muted tracks

cornil-m, 2019-11-28:
Would it be possible that the fill option doesn’t trigger tracks that are muted?

westlicht, 2019-12-01:
I guess this might be an option somewhere. I personally prefer having fill active on muted tracks too. It allows me to temporally bring in elements on muted tracks.

toneburst, 2020-02-02:
+1 for this, too. I was thinking exactly the same just the other day!

Could there be a setting somewhere to toggle current/suggested behaviour?

It could be a per-Track, or even per-Sequence setting, rather than a global thing.

westlicht, 2020-03-21:
Not sure I'd want it as a per track feature. It's one more option that probably is in the way most of the time. A global setting could make sense if I introduce a new settings page. See #85

westlicht, 2020-05-01:
Added _Fill Muted_ option to note tracks. Will be released with 0.1.37.

commit 6ea20458 (2020-05-01): added fill muted option

fixes #161
closed 2020-05-01 by westlicht