westlicht/performer #170: Midi input clarification / documentation

mike-kelly, 2020-01-23:
My main question is whether it is possible to input notes of sequences using an external midi controller (e.g. keyboard). The current mode (the one I'm aware of) of selecting the note then turning the encoder is not optimal for quick and intuitive work, IMO.
If that is not supported, is it possible in a future update?

Next question - I don't have a launchpad device, but I have tried connecting an Arturia keystep and it wasn't recognised. I also have an MPC Touch which is configurable as a midi controller - if that is recognised I suppose I would be able to configure it to behave like a Launchpad. Is there a reference for the midi signals communicated by the Launchpad?

Finally, it would be useful to clarify what differences there are, if any, between usb midi in and TRS midi in.

Thanks!

westlicht, 2020-01-23:
There are many ways to use a MIDI controller for inputing note data (see https://westlicht.github.io/performer/manual/#appendix-recording):

- select note layer, hold any number of step keys and press a key on the MIDI controller to assign the note
- use step recording: select _Step Record_ mode on project page, go to your sequence, arm recording and enter MIDI notes
- use live recording: select _Overdub_ or _Overwrite_ mode on project page, select a track, arm recording and record

All of the above also work with a Gate/CV source when activated on the project page.

You cannot really use the MPC Touch as a Launchpad replacement. For that you would have to implement a new _Controller_ in the firmware that implements the specifics. Just get a supported second hand launchpad if you want to use this mode.

There should not be any differences between USB MIDI and TRS MIDI. The only difference is that Launchpad controllers will be auto-detected and used in _controller mode_ instead of as a standard MIDI controller, so you cannot use it for doing routings for example.

closed 2020-01-23 by westlicht

mike-kelly, 2020-01-23:
Thanks for the clarifications, I missed that section of the manual.