westlicht/performer #172: Feature Request: Assign Midi Pressure to CV

mongrol, 2020-01-27:
As far as I can see Pressure is never actually used, or it is "mixed" with velocity to produce a single CV output on a midiCV track. Would be useful if we could assign it to a separate CV output thereby enabling aftertouch output to other modulation sources.

westlicht, 2020-03-21:
You should be able to get aftertouch output. To enable it you need to setup multiple physical CV outputs to get data from the same track (see _Layout_ page, read the manual).

westlicht, 2020-04-18:
Closing this as I assume it's working for you now @mongrol.

closed 2020-04-18 by westlicht

mongrol, 2020-04-18:
I don't get it. I set tracks 1 and 2 to both MidiCV tracks. I set Track one to "Pitch" only. All good. In Track 2 the options are only Pitch + Vel, or Pitch + Vel + Pressure. There's no way to only select pressure. Also, the values are only sampled (and then summed) at trigger. MIDICV tracks don't have an option to "Always update CV" like normal note tracks.

Also just tested using just note tracks. Track 2 is linked to Track 1's gate and CV. Track 2 is set to always update CV. But the CV from track 1 is only sampled on track 1's gate and thats what appears to be sent to Track2. So again, no varying CV from pressure.

westlicht, 2020-04-19:
You only need one track but assign multiple physical outputs to it on the _Layout_ page. You can have one MIDI/CV track output 4 notes of polyphony spread across all CV outputs as an extreme example. You can also have one track output pitch/velocity and aftertouch on 3 CV outputs and use the remaining 5 CV outputs for other tracks. It's very flexible.