westlicht/performer #173: Launchpad fill in Pattern Mode

toneburst, 2020-01-31:
Currently, there’s no way to trigger Fills from Launchpad in Pattern mode, though it is possible to do this in Sequence mode.

Could the method of triggering fills be copied over from Sequence to Pattern mode, please, as I think it would be really useful to be able to trigger fills here, without havin LG to switch Launchpad mode, or use the Per|Former hardware.

toneburst, 2020-01-31:
Can I also suggest it might be good to use one of the unused 1-8 (top) buttons in Sequence mode to enable a “momentary” feature, where patterns or scenes selected while this new modifier key is held will only play while the modifier is held, and the previously-playing pattern(s) / scenes will resume when the modifier is released?

westlicht, 2020-03-21:
> Can I also suggest it might be good to use one of the unused 1-8 (top) buttons in Pattern mode to enable a “momentary” feature, where patterns or scenes selected while this new modifier key is held will only play while the modifier is held, and the previously-playing pattern(s) / scene will resume when the key is released?

This can be achieved with the _next pattern_ fill mode. I don't want to add functionality to the Launchpad UI that is not part of the sequencer UI. So this is probably not happening.

westlicht, 2020-03-21:
> Could the method of triggering fills be copied over from Sequence to Pattern mode, please, as I think it would be really useful to be able to trigger fills here, without having to switch Launchpad mode, or use the Per|Former hardware.

I will check.

toneburst, 2020-03-26:
That’s great, thank you Simon.

westlicht, 2020-04-29:
Will be part of the 0.1.36 release.

commit 426a27b9 (2020-04-29): added fill and mute functions to pattern page

fixes #173
closed 2020-04-29 by westlicht