westlicht/performer #178: Feature request: track mutes in song mode

nickthekick, 2020-03-02:
a feature where you can add track mutes/no pattern in song mode would be great!

westlicht, 2020-03-21:
I think it's easy enough to just use an empty pattern, no?

nickthekick, 2020-03-21:
I’m using empty patterns now but for overview purposes it gets kind of messy. I think an x or - would make it way easier to see what parts that are muted. 

//Nick

westlicht, 2020-03-21:
I see. This feature would require some change to the data layout unfortunately as the pattern is stored in 4 bits now. I'll keep the issue open but it's not high prio

nickthekick, 2020-03-21:
Ok :)

//Nick

westlicht, 2020-04-18:
I have a partial implementation of this but did run into a bit of an issue. Performers song mode is also a performance feature to quickly chain some patterns. For this usecase one wants to control mutes manually. Now that song slots contain that information, it will basically set the slot's mutes when the slot is changed, overriding the current mute. So for example, if I quickly chain two patterns and mute one or several tracks, whenever the song slot changes the patterns change and the mutes are disabled. This is annoying. One option would be to check if there are any mutes set on a track across all slots in the song. If there are not, we simply don't set the mute state allowing song mode to be used as a performance feature. Once there is a single mute state defined on a track in the song, we always set the mute state when switching slots. But this still can be a bit unexpected I think. So not sure how to exactly tackle this.

nickthekick, 2020-04-19:
Yeah I understand. But if it’s mutes per pattern they could be linked right?  I’m thinking that it’s quite a fast workflow to duplicate pattern and mute what you want and duplicate again and change what you want. But maybe that’s my workflow only

//Nick

westlicht, 2020-04-29:
Will be part of 0.1.36 release.

closed 2020-04-29 by westlicht

nickthekick, 2020-04-29:
Sweet! Good work!

//Nick

westlicht, 2020-04-29:
Let me know how it works for you.

commit 3f0e6a21 (2020-04-29): added mutes to song slots

fixes #178