westlicht/performer #179: Song mode: song not starting properly

nickthekick, 2020-03-02:
If you the cursor is on line 2 and you press play (F button) the indicator on the left doesn’t show until you press play (F button) again. 

P.S shouldn’t the sequencer stop when you press stop on the F button? Don’t know if this is on purpose.

westlicht, 2020-03-21:
> If you the cursor is on line 2 and you press play (F button) the indicator on the left doesn’t show until you press play (F button) again.

I have to investigate, this might be a bug.

> P.S shouldn’t the sequencer stop when you press stop on the F button? Don’t know if this is on purpose.

Song mode is maybe a bit different to what it is in other sequencers. The play/stop button on the song page does engage/disengage the song mode. The idea behind is that it's usable in a live situation where you want to quickly chain some patterns together and play them in a loop maybe but then go back to jamming without stopping. I should probably rename that button to make it clearer.

westlicht, 2020-03-21:
> If you the cursor is on line 2 and you press play (F button) the indicator on the left doesn’t show until you press play (F button) again.

I can't reproduce. @nickthekick can you post a video or explain in more detail?

westlicht, 2020-04-18:
Closing due to lack of response I and me not finding anything suspicious.

closed 2020-04-18 by westlicht