westlicht/performer #181: Retriggers not evenly spaced across beats

sduck409, 2020-03-20:
Retriggers aren't currently divided evenly across the beat - they are clumped up near the beginning of the beat. For instance, if 2 retriggers is selected, the second trigger should happen in the middle of the beat - currrently that is not the case.

westlicht, 2020-03-20:
The retriggers are divided evenly. But only the ones that fit within the step length are triggered. Just increase the step length to the max and you get all retriggers. This allows to have _clumped_ retriggers as you call them, which is actually a feature I wanted not a bug.

sduck409, 2020-03-20:
Hi - Thanks for responding - yes, I think I figured it out, and yes it works well actually. Sorry for my misunderstanding!
-Steve

westlicht, 2020-03-20:
No worries, I think it is documented but its a bit of a hidden feature I guess,

closed 2020-03-21 by westlicht