westlicht/performer #183: LP mini MK2 doesn’t gte recognized

oxidom, 2020-04-07:
I just purchased a LP mini mk2 to use with the westlicht performer, I set the ID to 1 on the LP but the sequencer still doesn’t recognize it.. any ideas what could be wrong?

SloBloLabs, 2020-04-07:
Hi Oxidom, I had the same thing. It turned out one of the USB pins (D-) had a bad joint on the uC. Check connectivity between the USB connector and uC legs and resolder in case.

westlicht, 2020-04-12:
@oxidom the other thing to check is that you set the ID of the launchpad to zero. This affects the pid of the USB device and only the first one is detected by the sequencer.

closed 2020-04-12 by westlicht

oxidom, 2020-04-13:
thank you @SloBloLabs and @westlicht 
I tried re-soldering the usb port but no luck on that end I also tried to change the ID to 1 but same thing happens the LP is not recognized and goes into screensaver mode with blinking lights... the pid is 0x1036 and vid is 0x1235