westlicht/performer #184: Monitoring from CV input doesn't work.

mongrol, 2020-04-07:
Assign CV/gate to CV1/2 or 3/4 and pump in some CV from anywhere. Live monitoring the track doesn't play through to the track CV or gate outs. You can record CV input and gate input and write it to the track, but you can't play it from a CV keyboard like mentioned in the manual.

Dunk303, 2020-04-13:
You can play a track from external cv it as long the sequencer is stopped!

westlicht, 2020-04-18:
I can't reproduce the behaviour. Live monitoring works the same with CV/gate as with a MIDI controller. It works even if both sources are used at the same time. It also works when the sequencer is running. This is with regards to note tracks.

Currently you can't select CV/Gate as a _Source_ on a MIDI/CV track, only MIDI and USB MIDI can be selected. Is that what you mean? I think I might add that option.

Otherwise, can you please explain in more detail what you mean?

mongrol, 2020-04-18:
Yep, my mistake. I was inputting CV from a S&H driven by a noise source, which obviously would only change when the clock was running. I've tested CV monitoring from several other sources with both sequencer stopped and running and everything is working correctly. Can be closed. User error.

closed 2020-04-18 by westlicht