westlicht/performer #187: Track not updating on Overview screen when set to /4 Division

theotteryears, 2020-04-30:
Purely cosmetic issue - project file attached. 

[002.PRO.zip](https://github.com/westlicht/performer/files/4556968/002.PRO.zip)

westlicht, 2020-05-01:
Thanks for uploading. However, I just tested this on the simulator and everything looks fine. I will run it on the hardware as well, but I wasn't able to reproduce before, so I don't expect it to be different for this project. You are running the latest version, right?

theotteryears, 2020-05-01:
Yeah I just got 1.36 so I’ll try it there. 

Ok no worries! It’s only visual anyway so not a problem. Thanks for putting in the curve track cv thing. Works awesomely! 

Great work on this sequencer. It’s become my favourite already.

westlicht, 2020-05-01:
If there is an issue I'd very much like to fix it. So please give it another go. Maybe there is something else you do that I missed. But just loading the project and letting it play did not show the described behavior here.

theotteryears, 2020-05-01:
Ok I’ll try to recreate.

westlicht, 2020-05-13:
@theotteryears did you manage to reproduce?

theotteryears, 2020-05-13:
sorry! I haven't had chance yet, but will do this tomorrow for you.

theotteryears, 2020-05-13:
@westlicht Hi, yeah I've tried and cannot recreate this. not only that, but the project which had the problem now longer has the problem!! so weird, but I will close this as there's no issue, sorry to waste your time :)

closed 2020-05-13 by theotteryears