westlicht/performer #188: Option to fire the RESET OUT slightly later than the last clock pulse

theotteryears, 2020-05-04:
I have some sequencers which I clock from Performer which do not react well to the current way the reset out works which seems to fire at the same time as the last clock pulse out when pressing the start/stop button. it would be useful to have an option to disable the reset out getting fired when the sequencer is stopped, or be able to delay the reset out slightly on sequencer stop.

or have a new button combo which you could manually send a reset out from. Hope this makes sense. I'm having trouble getting my Trigger Riot synced up with this properly as it resets to the first step when I stop Performer and then its always one step in front when I start Perfomer again.

westlicht, 2020-05-31:
If you don't want a reset/run signal, you could just not connect it. An option would be to add a _Disabled_ mode to _Output Mode_ on the clock page. But without the manual reset feature that would be rather pointless. Or do I miss something here?

theotteryears, 2020-05-31:
Yeah don’t worry! I mean to close this sorry

closed 2020-06-02 by theotteryears