westlicht/performer #190: FEATURE REQUEST? Midi polyphony

kurtjoy, 2020-05-10:
Apologies if this has been bought up before..I did a search but couldnt find anything.
Is it possible to trigger two midi notes at once. Im thinking mainly for drum programming, where two drums hits would hit at one time. Maybe integrate it like the secondary curve or gate has been added to curve tracks.
Performer is my primary sequencer, so trying to squeeze the maximum out of each track as possible!

westlicht, 2020-05-12:
No there is no such option. Performer is primarily a CV/Gate sequencer, not a MIDI sequencer. It would mean to add a new gate layer to the note track, but then it would also need all the other layers for all the probability, gate length etc. stuff. So essentially a second track.

ronnyjamesdisco, 2020-05-15:
How about polyphony just as a MIDI thru option? 

I´d love to use my keyboard that´s already connected to PER|FORMER on a dedicated channel to play something downstream the MIDI-out polyphonically - even when not being able to process this in the sequencer itself. Just thinking out loud ;)

westlicht, 2020-06-14:
Pass through might be part of extend configuration options for MIDI handling. Will give it a thought.