westlicht/performer #191: Feature Request: MIDI data input

gitmusik, 2020-05-11:
I wonder if anyone else would find this a useful feature?

A special mode whereby an attached MIDI controller with, say, 16 rotaries, could control whatever data the screen is currently displaying using MIDI Note Numbers or CCs.
This would be great for live performance.
For example...if the first Note page was selected Rotary 1 would adjust the note value of Note 1 (according to all the defaults of scale and so on).
...and Rotary 16 would be able to adjust the same on Note 16.
(This mode would need to be able to recognise the controllers 1-16 when the screen is displaying Steps 17-32, 33-49, etc to keep the controller of a manageable size.

The same for all the other adjustable data points....probability, length, curve, gate etc...
I do not know whether it would need to be a latching or instantaneous recognition.
It may be impossible to implement, I have no idea.

This would be so much faster than the present method, good though that is, and would make the BEST sequencer in the world the best in the universe. By a mile. Alright, a light year!

westlicht, 2020-05-12:
This was discussed before, see #64. I was and still am not a big fan of it. Please read through the comments there and share your opinion int this issue here.

gitmusik, 2020-05-13:
Thank you for your reply, Westlicht, and my apologies for not having seen the earlier thread.
I shall investigate the Launchpad implementation more fully.

closed 2020-05-13 by gitmusik