westlicht/performer #192: feature request: output C1 to all CV outputs for tuning VCOs

captainhooke88, 2020-05-13:
soon to be a new user here and coming from the eloquencer, there are a couple of features on the eloquencer i wish would be on the per|former which might be interesting to consider.

one of them being a utility to output a certain note over all CV outputs. very convenient when you have multiple VCOs hooked up to the sequencer and you don't want to go through the hassle of going in each track and outputting a note. a small but very time efficient tool i think :)

westlicht, 2020-05-13:
You can do that on the calibration page, I personally use that all the time to tune VCOs. But maybe there indeed could be another less hidden option for it.

captainhooke88, 2020-05-13:
ah cool, for me can be as hidden as possible as it is something you only use once when firing up everything but that the option is already there is very nice.

closed 2020-05-13 by westlicht