westlicht/performer #193: Feature request: octave jump when pressing + rotating the encoder

thruzero, 2020-05-13:
The actual method to jump between octaves when editing notes is with the combo 'shift button + step button + rotate encoder'. It'd be nice if the same could be achieved as well with 'step button + encoder button + rotate encoder'. That way wouldn't be necessary to use one extra finger for the shift button. It'd be handier/faster at least for me!

<3 per|former

westlicht, 2020-05-14:
The encoder press is used to toggle gates even if in note layer. That is a really important feature for me as I can enter notes and gates on an empty sequence both at the same time.

thruzero, 2020-05-18:
Ooooh true, true, now I remember it from the manual but I forgot! Also didn't notice on the screen that the step square got filled when the encoder was pressed. Still used to other sequencers with that workflow. I should set gates only from note page from now on as punishment :)

closed 2020-06-14 by westlicht