westlicht/performer #195: Gate/Note trigger on stop?

PlanetZaxxon, 2020-05-23:
Hello! So I recently just picked up a Performer and each time I hit stop, to stop the sequencer, a note/gate gets triggered, and there are notes that don't even exist anywhere in the sequence. I just can't seem to figure out why it would be doing that. I just updated to the latest firmware and it didn't seem to help.

Any thoughts?
Thanks!

here is a video of the issue: https://drive.google.com/open?id=1D-J2joniDMv-otrNW0VkTMbIFz2vc3iB

PlanetZaxxon, 2020-05-23:
Some more information:

Even stranger, when i initialize the sequence and  just put in gates, which by default it triggers C0 at every gate, and then when I hit stop it makes no extra trigger! Then I change the note values on the gate, now when I hit stop, it triggers some note/gate on stop, the issue I have been describing. So then I can initialize the sequence again, put in some gates only, and I hit stop and no extra trigger, its completely silent, as it should be. It's only after I start changing the note values, does the stop issue arise. Driving me nuts over here haha...

westlicht, 2020-05-23:
It definitely should not output a trigger/gate. But it will reset the voltage to 0V. I think I had a good reason for doing that, but I might have to rethink that reset to 0V. It never bothered me (or anybody else it seems). Typically you don't stop the sequencer all the time :)

PlanetZaxxon, 2020-05-25:
Ok coo, thanks for the repsonse! I will be on the lookout if any new firmware changes. :)

theotteryears, 2020-06-03:
This is a bit of a problem at the end of the song when everything drops to 0v and sounds weird :) I'd like it to hold the last value if possible on the sequencer stop.

rdomain, 2020-10-18:
Yep, I'd like this also.  I've been working on a performance where the sequencer starts and stops throughout the performance and I have to be careful not to trigger any bum notes when I stop it running.  That would be awesome.

> This is a bit of a problem at the end of the song when everything drops to 0v and sounds weird :) I'd like it to hold the last value if possible on the sequencer stop.

westlicht, 2020-10-22:
I still have to think about this for a bit. But there is a workaround you can use in the meantime. On clock setup you can specify what action the `SHIFT` + `PLAY` combo does. By default it is set to _Restart_, meaning the clock will instantly restart. You can switch this to _Pause_, which allows you to pause/unpause the clock. If the sequencer is paused, you can still hit `PLAY` to start from the beginning. So just configure to _Puase_ mode and then use `SHIFT` + `PLAY` to stop and just `PLAY` to start.

rdomain, 2020-10-23:
> I still have to think about this for a bit. But there is a workaround you can use in the meantime. On clock setup you can specify what action the `SHIFT` + `PLAY` combo does. By default it is set to _Restart_, meaning the clock will instantly restart. You can switch this to _Pause_, which allows you to pause/unpause the clock. If the sequencer is paused, you can still hit `PLAY` to restart from the beginning. So just configure to _Pause_ mode and then use `SHIFT` + `PLAY` to stop and just `PLAY` to start.

Works a treat!  Thanks for that.  :)

rdomain, 2020-11-19:
Now that's I've got a Launchpad Mini Mk3 up and running, this little workaround doesn't work via the Launchpad unfortunately.  Would it be possible to have a variation of the play/stop of 8+7 button combo....   Have the original 'shift + play' work around become a 7+8 button combo on the Launchpad and the standard play/stop is still 8+7?

Apart from this, the Launchpad integration is great!  Thank you.