westlicht/performer #196: Configurable frame for shifting steps

SloBloLabs, 2020-05-26:
The first time today I really needed the shifting steps feature, because I played some pattern in some weird offset within the beat. Interestingly, the step shifting is always applied to the whole range of 64 steps. I suppose configuring the shifting range (e.g. between first step and last step) would make it more intuitive. It was my use case in that very moment, at least.
I know you can achive this with rotation applied to the whole track, of course. But I like to have my patterns prepared in lets say a clean way, and typically, the rotation for one pattern wouldn't match for others. Messing things up with step rotation is something I would only play with occasionally knowing that zero rotation brings me back to safe house at any time.
Let me know if I miss something here. :-)

westlicht, 2020-05-31:
It's something I have previously thought about myself. It might be more intuitive if steps are shifted in the current first/last step range. I will consider changing the behaviour.

Lokua, 2020-07-10:
I would love this - it is what I expected the first time I tried to shift steps.

This is perhaps scope creep and technically a different feature request but complimentary and related: 

It would be amazing if shifting steps worked with respect to selections. So with no selected steps, it would be exactly what @SloBloLabs mentioned, however when steps are selected, holding shift and pressing next would turn for example `1:A,3:C,5:E` into `1:E,3:A,5:C`, leaving any other steps outside of 1,3,5 unaltered. This would be analogous to how transposing octaves works so seems like it would fit into the existing Performer UX quite naturally.

westlicht, 2020-07-19:
I think I'll add two changes:
1. Shift steps in the current first/last step region as requested by @SloBloLabs if there is no step selection.
2. Shift selected steps as mentioned by @Lokua if there is a step selection.

westlicht, 2020-07-20:
Will be part of the 0.1.37 release.

closed 2020-07-20 by westlicht

Lokua, 2020-08-01:
WOW! Thank you!

commit 0d8b04f0 (2020-07-20): improved step shifting

fixes #196