westlicht/performer #197: Midi Input Channel

captainhooke88, 2020-05-27:
I swear to god I did this before, but I either completely lost how to do it or it was never an option.

either way;

is it possible to change the midi channel the midi input listens to when inputting notes? or turn of listening to note information and only listen to clock information or something?

my situation:

my octatrack is sending out midi to different midi devices and the performer in its turn gets midi from a thru from one of my synths. I want to setup midi channels so for instance the performer listens to midi channel 5 only, so my synths and other gear can listen to other midi channels.

captainhooke88, 2020-05-27:
or I could commit certain tracks to certain midi channels. for instance midi 1 to 8 send note info to track 1 to 8 at all times and ten 9 to 16 I could use for my other gear.

westlicht, 2020-05-31:
That feature has been requested many times now, I will have to implement it. Currently there is no way to restrict the midi port/channel which is used for monitoring (live playing/recording notes on a sequence). However, all events that are consumed for routing purposes or for a MIDI/CV track are not sent to monitoring. That might allow you to setup a temporary workaround, but it's far from idea.

captainhooke88, 2020-05-31:
Ill have a look into that workaround :) thanks and great to hear it’s something to look forward to.

captainhooke88, 2020-06-08:
I got this to work by sacrificing one channel on the performer to a certain midi channel, but I am wondering if it's possible to do it with routing as well. I prefer to have all 8 tracks available for sequencing and the routing I use less.

but if I set up a routing for midi note information on midi channel 1 for instance, it does not exclude note information from monitoring currently active track from that midi channel? am I missing something or is that not what you meant?

captainhooke88, 2020-06-30:
hi Simon, just wondering if you have any idea when you might be able to update the firmware with this functionality..? Im on my knees and it would be amazing to have this possibility, my setup is really craving for it.

#daretoask

westlicht, 2020-07-20:
I'm looking into this right now. Should be coming with the next update.

closed 2020-10-19 by westlicht

westlicht, 2020-10-22:
Will be released with 0.1.38.

commit c4cb3397 (2020-10-19): added midi input option to project

fixes #197